HViktorTsoi/FAST_LIO_LOCALIZATION #1: fialed installing open3d

ZeyingXuHuaWei, 2021-08-11:
anyone succeed in installing open3d with python2.7?

HViktorTsoi, 2021-08-11:
Please post the specific behavior of your issue

ZeyingXuHuaWei, 2021-08-11:
root@ubuntu:# pip install open3d==0.9 -i https://pypi.tuna.tsinghua.edu.cn/simple
Collecting open3d==0.9
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/67/6e/4b0b124007f642c6305b4444ed7c143867d3fc74e90efaf0c51ecfe5db6a/open3d-0.9.0.0-cp27-cp27mu-manylinux1_x86_64.whl
Requirement already satisfied (use --upgrade to upgrade): widgetsnbextension in /usr/local/lib/python2.7/dist-packages (from open3d==0.9)
Requirement already satisfied (use --upgrade to upgrade): ipywidgets in /usr/local/lib/python2.7/dist-packages (from open3d==0.9)
Collecting notebook (from open3d==0.9)
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/77/2f/b662a3f6306ef2f494f33a4a632c34dcd012568e9ee2dbd985cfb1429105/notebook-6.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-8_LzK2/notebook/setup.py", line 42
        print(error, file=sys.stderr)
                         ^
    SyntaxError: invalid syntax

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-8_LzK2/notebook/

After  i set install notebook==5.7 , then I get the following error

[image]

ZeyingXuHuaWei, 2021-08-11:
It seems that I need to upgrade my pip version.    There is another problem. If I install open3d 0.9 and import open3d, then  I will get error from .open3d import * # py2 py3 compatible . This error does not happen if  I install 0pen3d 0.7

HViktorTsoi, 2021-08-11:
> It seems that I need to upgrade my pip version. There is another problem. If I install open3d 0.9 and import open3d, then I will get error from .open3d import * # py2 py3 compatible . This error does not happen if I install 0pen3d 0.7

We have tested successfully on **Open3D 0.8 & 0.9**, but haven’t tested on **0.7**. According to the documentation, 0.7 ~ 0.9 have consistent APIs, therefore **0.7** is theoretically feasible. Please let us know if you test successfully on **0.7**, we will update the dependencies.

HViktorTsoi, 2021-08-11:
> It seems that I need to upgrade my pip version. There is another problem. If I install open3d 0.9 and import open3d, then I will get error from .open3d import * # py2 py3 compatible . This error does not happen if I install 0pen3d 0.7

Please post the full error log of `from .open3d import * # py2 py3 compatible`, it seems like issue with the python version

HViktorTsoi, 2021-08-11:
> It seems that I need to upgrade my pip version. There is another problem. If I install open3d 0.9 and import open3d, then I will get error from .open3d import * # py2 py3 compatible . This error does not happen if I install 0pen3d 0.7

Add **Open3d 0.7** support, please pull the latest code.