HViktorTsoi/FAST_LIO_LOCALIZATION #4: Localization Problem when Lidar is not moving over long period of time

David9696, 2021-08-16:
Hi,

We found a problem when the livox lidar was not moving over a long period of time during localization mode. This is the error:

[WARN] [1629076036.379140]: Not match!!!!
[WARN] [1629076036.380224]: [[ -1.58418010e-02   5.23801209e-02   9.98501558e-01   7.28467931e+01]
 [ -2.20726244e-02   9.98365197e-01  -5.27231625e-02   1.47314253e+00]
 [ -9.99630850e-01  -2.28747797e-02  -1.46597361e-02  -2.55931770e+01]
 [  0.00000000e+00   0.00000000e+00   0.00000000e+00   1.00000000e+00]]
[WARN] [1629076036.380964]: fitness score:0.0
[INFO] [1629076036.632107]: Global localization by scan-to-map matching......
[Open3D WARNING] [KDTreeFlann::SetRawData] Failed due to no data.
[Open3D WARNING] [KDTreeFlann::SetRawData] Failed due to no data.
[INFO] [1629076036.923809]: Time: 0.290438890457
[INFO] [1629076036.924859]: 

The Odometry topic was empty. The Localization topic was publishing very big number in x,y and z.

Do you know what is the problem?

Thanks.

HViktorTsoi, 2021-08-16:
Hi, did the localization module initialize successfully? And how long did you place LiDAR without moving?

We observed from your log that, the **fitness score** is 0, and there is a warning says '[KDTreeFlann::SetRawData] Failed due to no data.', this is usually because the scan published by fast-lio is empty, or the offline map is empty; conserning that the Odometry topic is also empty, it may be because fast-lio crashed.

Here are some debugging advices:
1. First make sure the system is initiallized sucessfully (This information will show in the terminal log);
2. Run [FAST-LIO](https://github.com/hku-mars/FAST_LIO) independently, under your test scene, staying the LiDAR still for a long period of time, and see if the fast-lio will crash (not publishing the **Odometry** topic or **cloud_registered** topic ). If it crashes, please post the error log here;
3. If possible, it will be more efficient if you may provide the test bag and the sample map for debugging, to cuijiahetony@live.com .

David9696, 2021-08-16:
Hi,

Yes, the localization module was initialized successfully. The lidar was not moving for about 3 to 4 hours.

Yes, the Odometry topic was empty. And the fast-lio node crashed. This is the error shown when it crashed:

Leaf size is too small for the input dataset. Integer indices would overflow.[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.

Hope the information above is enough for you to debug.

In the meant time, I would try to record the bag file. But, the recorded bag file would be quite big as the the problem would only appear when the lidar is not moving for 3 to 4 hours.

HViktorTsoi, 2021-08-16:
> Hi,
>
> Yes, the localization module was initialized successfully. The lidar was not moving for about 3 to 4 hours.
>
> Yes, the Odometry topic was empty. And the fast-lio node crashed. This is the error [...]

Then the issue may be clear: it is because the odometry estimated by fast-lio drifted away for a very long distance when the LiDAR was not moving for a log time, and failed to provide a feasible initial solution for the localization module. Therefore the localmap maintained by fast-lio is so large that the **Integer indices** of the voxel filter overflowed, and this may also explain why the localization module publishes very large number(because the odometry value itself is abnormally large).

When the LiDAR is not moving, pleasure make sure there is enough point clouds observed by the LiDAR. For example, if you put the LiDAR directly facing a wall, then it can only observe a surface, there is not enough constraint from either sides, then the odometry will drift to the sides in a long period of time. 

Also it's best to post this issue to [FAST-LIO](https://github.com/hku-mars/FAST_LIO/issues) because it is essentially an abnormally-odometry-drifting problem(unlike odometry drifting caused by the accumulative error, which can be eliminated by relocalization), there may be some bug-fix in the future version of fast-lio to tickle this problem.

David9696, 2021-08-16:
Hi,

Yes, you are right. When the lidar was not moving, it was facing a wall with a door. Seems like the problem might be from the Odometry estimation. I have logged an issue at FAST-LIO GitHub. But no solution is given yet. Maybe should halt the Odometry estimation when the lidar is not moving?

Anyway, thanks for your great help.