HViktorTsoi/FAST_LIO_LOCALIZATION #5: using offline map for odometry

ZeyingXuHuaWei, 2021-08-16:
Hi author, I think you only use offline map for global localization, can we also use offline map to replace the online map for odometry?  Do you think this a good idea?

HViktorTsoi, 2021-08-16:
Yes, actually we are working on this, the current version is a pretty 'loosly coupled', we only use the odometry and deskewed-scan from fast-lio. Infact, it's possibly to replace the local map maintained by fast-lio to the offline map, and since fast-lio-2 use a direct-matching method without feature extraction, the implementation may aslo be intuitive.

ZeyingXuHuaWei, 2021-08-16:
Yes.  With very little modification I can now use offline map as the local map of fast-lio.

closed 2021-08-16 by ZeyingXuHuaWei

siyuw-rf, 2024-08-14:
> Yes. With very little modification I can now use offline map as the local map of fast-lio.

Hi, I am very interested in such feature, do you mind sharing the modification?