HViktorTsoi/FAST_LIO_LOCALIZATION #8: Python2 can not install open3d due to pyrsistent requiring python > 3.5

miku39s3, 2021-08-28:
I have used pip2 and pip3 to install pyrsistent==0.15 both in python2 and pyhton3. But when I install open3d=0.9, it still shows a error that pyrsistent requires Python '>=3.5' but the running Python is 2.7.17.

HViktorTsoi, 2021-08-28:
> I have used pip2 and pip3 to install pyrsistent==0.15 both in python2 and pyhton3. But when I install open3d=0.9, it still shows a error that pyrsistent requires Python '>=3.5' but the running Python is 2.7.17.

Please try `pip install open3d-python`

xuzhen80, 2021-09-06:
> I have used pip2 and pip3 to install pyrsistent==0.15 both in python2 and pyhton3. But when I install open3d=0.9, it still shows a error that pyrsistent requires Python '>=3.5' but the running Python is 2.7.17.

Has your problem been solved？I also encountered this problem and would like to ask for your help。

HaisenbergPeng, 2021-09-10:
> pip install open3d-python

Not working...

ZeyingXuHuaWei, 2021-09-10:
Try:
pip install open3d==0.9  pyrsistent==0.15

HaisenbergPeng, 2021-09-10:
the same error

HaisenbergPeng, 2021-09-10:
Here is my full solution for your reference
a)PROB: RuntimeError: Python version >= 3.7 required：numpy版本过高 
pip install open3d-python numpy==1.16.6
b)PBOB: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-x0d0hwhj/pyzmq/
sudo python -m pip install --upgrade --force pip 
sudo pip install setuptools==33.1.1
c)PROB: pip is no longer available：bash: /home/binpeng/.local/bin/pip: No such file or directory
d)hash -r
e)pip install open3d==0.9 numpy==1.12 pyrsistent==0.15