HViktorTsoi/FAST_LIO_LOCALIZATION #9: fatal error: fast_lio/Pose6D.h: No such file or directory

HaisenbergPeng, 2021-09-10:
I was wondering why no one encounters such an error.
Here is how I solved it: change the package name in common_lib.h from "fast_lio" to "fast_lio_localization".

PanNeng-ZJU, 2022-02-21:
It helps, thank you!

srinivasrama, 2022-04-14:
@WonderSimiliar you need to change all `fast_lio` to `fast_lio_localization`

JustThinker, 2022-04-14:
> @WonderSimiliar you need to change all `fast_lio` to `fast_lio_localization`

your method works! thanks a lot! there are only two "fast_lio'' that should be replaced by 'fast_lio_localization'

WonderSimiliar, 2022-04-28:
> @WonderSimiliar you need to change all `fast_lio` to `fast_lio_localization`

it works! Thanks!

Salon-dot, 2023-12-14:
I have changed all，but it also doesn‘t work

Salon-dot, 2023-12-14:
fast_lio_localization/Pose6D.h: No such file or directory

XC-CN, 2025-02-21:
Don't forget to do in common_lib.h
// 修改第36行的命名空间
typedef fast_lio_localization::Pose6D Pose6D;  // 原为fast_lio::Pose6D