HViktorTsoi/FAST_LIO_LOCALIZATION #13: Rivz running positioning algorithm is very laggy, is there a way to fix it?

LiquoriceH, 2022-07-08:


sangminkim-99, 2022-07-20:
Hi,

It might be hard to execute the localization and visualization at the same time for your virtual machine.
You can run the localization algorithm without any visualization, and record the desired output with `rosbag record <some topic you want to record>`.
Then turn on the rviz, and visualization with the recorded output with `rosbag play <your bag file>`.

I hope that this method relieve your problem...

Thank you.

LiquoriceH, 2022-07-22:
> Hi,
>
> It might be hard to execute the localization and visualization at the same time for your virtual machine. You can run the localization algorithm without any visualization, and record the [...]

Realllly thanks！