HViktorTsoi/FAST_LIO_LOCALIZATION #16: "Not Matching" means fail to localization?

caomengnb, 2022-09-16:
My terminal shows like follow：

[INFO] [1663312150.392011]: Global localization by scan-to-map matching......
[INFO] [1663312151.285043]: Time: 0.88362288475
[INFO] [1663312151.286561]: 
[WARN] [1663312151.287811]: Not match!!!!
[WARN] [1663312151.289611]: [[ 0.98088933  0.19430204  0.01014116  7.2997208 ]
 [-0.1944272   0.9808267   0.01330566 -0.4114994 ]
 [-0.0073614  -0.0150231   0.99986005  0.54942485]
 [ 0.          0.          0.          1.        ]]
[WARN] [1663312151.290827]: fitness score:0.364882312047
[WARN] [1663312151.292033]: Waiting for initial pose....

Fitness score is only 0.364882312047. It seems fail to localization. How to solve this problem?

caomengnb, 2022-09-16:
In addition, my lidar is velodyne. I constructed a map by FAST-LIO. I use the same map and rosbag to localize by FAST_LIO_LOCALIZATION, but only receive "Not match!!!!".

cjcjcj7777, 2022-09-26:
Hi I am using ouster lidar and map built using fast_lio. 
1) When I test fast_lio_localization using a bag file with which map was built, fast_lio_localization works without any error
2) When I test fast_lio_localization in real-time, fast_lio_localization orients lidar_data with origin of map, when I give initial pose w.r.t to it's current position in map in rviz it throws warning "Not match!!". The same occurs when I publish initial pose using the following command 
`rosrun fast_lio_localization publish_initial_pose.py 14.5 -7.5 0 -0.25 0 0  `

Please let me know how can I rectify the issue.

arjunskumar, 2022-09-26:
@cjcjcj7777 you should give your initial pose after `rosrun fast_lio_localization publish_initial_pose.py`, also try to check your fitness score and relax the threshold

wozhishiyikedan, 2023-03-08:
I solved it. If you use mid360, you need to set FOV to 360 in global_localization.py

DoongLi, 2023-08-19:
@wozhishiyikedan Hi，I also use Mid360，should I modify global_localization.py or Avia.yaml?

xjtulixuanyi, 2023-10-09:
> @wozhishiyikedan Hi，I also use Mid360，should I modify global_localization.py or Avia.yaml?

global_localization.py

DoongLi, 2023-10-09:
@xjtulixuanyi  Thx

66Lau, 2023-10-22:
> I solved it. If you use mid360, you need to set FOV to 360 in global_localization.py

It really works, thx!