HViktorTsoi/FAST_LIO_LOCALIZATION #28: Ros2 Support

JaYNits, 2023-07-17:


Taeyoung96, 2023-07-20:
@JaYNits 
Although not based on this repository,  
I've been experimenting with a LiDAR localization called hdl-localization that works in ros2.  
https://github.com/Taeyoung96/hdl-localization-ROS2

JaYNits, 2023-08-16:
@Taeyoung96 Thanks, does HDL Support Livox Mid360?

Taeyoung96, 2023-08-17:
@JaYNits   
Unfortunately, only Velodyne is currently implemented. :cry:

JaYNits, 2023-09-25:
@Taeyoung96 if i provide you with the bag files of Livox Mid360 , can you test it ?

JaYNits, 2023-09-25:
@HViktorTsoi Any Plans for ros2 ?

Alexbeast-CN, 2023-12-18:
The ROS1 version of hdl_localization works well for Livox MID360.

laksh-nanwani, 2025-03-26:
@JaYNits @Taeyoung96  @Alexbeast-CN @XW-HKU @borongyuan 
Checkout - [FAST-LIO-LOCALIZATION2](https://github.com/Smart-Wheelchair-RRC/FAST_LIO_LOCALIZATION2)
I have modified the files to work with ROS2 and tested with Livox MID360. It would be great if you guys could try it out with other Lidars and let me know how they work and raise issues, if any.