HViktorTsoi/FAST_LIO_LOCALIZATION #32: The issue of dependency installation for open3d packages

Rax0913, 2024-01-08:
When I run ''pip install open3d==0.9" in cmd, there is a error.
"ERROR: Could not find a version that satisfies the requirement comm>=0.1.3 (from ipywidgets->open3d==0.9) (from versions: 0.0.1)
ERROR: No matching distribution found for comm>=0.1.3 (from ipywidgets->open3d==0.9)"
It seems that the 0.1.3 version of comm cannot be installed in the Python 2.7 environment, I have also tried the "pip install comm==0.1.3",  but it still reported an error.
 "ERROR: Could not find a version that satisfies the requirement comm==0.1.3 (from versions: 0.0.1)
ERROR: No matching distribution found for comm==0.1.3".
How can I  solve this problem?

SamoaChen, 2024-01-16:
it seems like ipywidgets version sorted by pip to install open3d==0.9 is too new. install an older verion as shown below will resolve the issue 
_pip install ipywidgets==7.6.3_

ga58lar, 2024-09-14:
See https://github.com/HViktorTsoi/FAST_LIO_LOCALIZATION/issues/43#issuecomment-2351068081:
`pip install pyrsistent==0.15 open3d==0.9 --no-deps`