HViktorTsoi/FAST_LIO_LOCALIZATION #36: Error in visualizing the map

proxi666, 2024-03-04:
Hello 
I have successfully created a map using Fast_lio.
When I was working with Fast_lio_localization, I am unable to see the map in Rviz, but the localization through ROSBAG is working (check SS) please do provide some help on how can I make it work...

[image]

[image]

proxi666, 2024-03-04:
After giving initial pose, it says 
_first scan not received_

fbaygul, 2024-11-12:
you have to play a bag or provide pointcloud data to the system

RuifengLu, 2024-12-17:
same problem, how did you solve it?

fhawk-de, 2024-12-23:
I solved this problem by:
In ROS Noetic and later versions, the slash (/) before the frame ID may cause issues.
In the launch file,set frame_id to 'map' instead of '/map'.