HViktorTsoi/FAST_LIO_LOCALIZATION #41: 实时定位很慢请问有解决办法吗

pl9632008, 2024-07-19:
实时定位很慢请问有解决办法吗

arjunskumar, 2024-07-19:
@pl9632008 share your config

pl9632008, 2024-07-22:
Hello, I found that fastlio keeps adding ikdtree nodes. I added the following code in laserMapping.cpp to delete nodes at intervals. Additionally, I modified filter_size_surf to 1 and filter_size_map to 1. After that, I found that localization works. Will this cause any potential risks? The lidar is Livox HAP.   
            int max_len;
            nh.param<int>("max_len", max_len, 10000);
            if(featsFromMapNum>max_len){
                BoxPointType cus_range = ikdtree.tree_range();
                std::vector<BoxPointType> res={cus_range};
                ikdtree.Delete_Point_Boxes(res);
                
            }

pl9632008, 2024-07-22:
Hello, I found that fastlio keeps adding ikdtree nodes. I added the following code in laserMapping.cpp to delete nodes at intervals. Additionally, I modified filter_size_surf to 1 and filter_size_map to 1. After that, I found that localization works. Will this cause any potential risks? The lidar is Livox HAP.   
     ```
       int max_len;
            nh.param<int>("max_len", max_len, 10000);
            if(featsFromMapNum>max_len){
                BoxPointType cus_range = ikdtree.tree_range();
                std::vector<BoxPointType> res={cus_range};
                ikdtree.Delete_Point_Boxes(res);
            }
```