HViktorTsoi/FAST_LIO_LOCALIZATION #43: Did anyone install open3d0.9 successfully ?

charon-cheung, 2024-09-03:


ga58lar, 2024-09-14:
If you want to follow the instructions use:  
```
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py && \
    python2.7 get-pip.py

alias python=/usr/local/bin/python2.7

pip install --upgrade pip setuptools wheel
pip install --upgrade setuptools --user python
pip install pyrsistent==0.15 open3d==0.9 --no-deps
```

But I highly recommend to use the fork of Koide: https://github.com/koide3/FAST_LIO_LOCALIZATION, as he already provides a Dockerfile for the build.

championAI, 2025-10-17:
> 通过安装低版本的ipywidgets实现安装open3d 0.9版本，但是在ubuntu20.04中我运行时遇到了以下问题 File "/home/jgdog1/FASTLIO2/src/FAST_LIO_LOCALIZATION/scripts/transform_fusion.py", line 10, in import rospy File "/opt/ros/noetic/lib/python3/dist-packages/rospy/**init**.py", line 49, in from .client import spin, myargv, init_node, File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 60, in import rospy.impl.init File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/init.py", line 54, in from .tcpros import init_tcpros File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros.py", line 45, in import rospy.impl.tcpros_service File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 54, in from rospy.impl.tcpros_base import TCPROSTransport, TCPROSTransportProtocol, File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 167 (e_errno, msg, *_) = e.args ^ SyntaxError: invalid syntax

你好，请问你这个问题解决了吗，我也遇到同样的问题。

[image]

junxiao13, 2025-10-17:
> 你好，请问你这个问题解决了吗，我也遇到同样的问题。
> 
> <img alt="Image" width="1364" height="464" src="https://private-user-images.githubusercontent.com/72303631/502345019-9e9e0b8b-372d-4232-9924-1799a59bcfb5.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjA2ODI3NTUsIm5iZiI6MTc2MDY4MjQ1NSwicGF0aCI6Ii83MjMwMzYzMS81MDIzNDUwMTktOWU5ZTBiOGItMzcyZC00MjMyLTk5MjQtMTc5OWE1OWJjZmI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEwMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMDE3VDA2MjczNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkYTBkOGIwZmE1NjRlNzJkNzI3ZGI2YjEzMjlkYmM2Mjg3MDk2YjQ1YjBhZjk2ODg3ZmFmMGZmNDZhZTI1ZTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.jsLe9kLZOBEHbTn1U8yWV-AEAjnY9e2YAYJnps2tJww">

之后我在提交界面里看到了其他开发者提供的修改, 在我这边测试是可行的，对应的提交为https://github.com/HViktorTsoi/FAST_LIO_LOCALIZATION/pull/20 ，可以参考其中的内容修改FAST-LIO-LOCALIZATION的全局定位模块, 不用安装Python2环境，直接用系统自带的Python3即可，也就不会因为解释器版本导致库和源代码出现冲突.