HViktorTsoi/FAST_LIO_LOCALIZATION #44: process has died

Parkeunseok, 2024-09-05:
After executing the command, the following error appears.

command : roslaunch fast_lio_localization localization_ouster64.launch map:=/home/a/main_ws/fast_ws/src/FAST_LIO_LOCALIZATION/PCD/VECtor_school.pcd

[image]

What is the solution?

Parkeunseok, 2024-09-05:
The above error has been resolved

However, how do I resolve this error?

[image]

And the map is not clearly visible as in the example.

[image]

How do I solve this problem?

QingKe-UP, 2024-09-24:
@Parkeunseok 
Hello, how did you solve the first problem?
Looking forward to your answer.

Parkeunseok, 2024-09-26:
@QingKe-UP  Hi, the first issue was when I tried to run the code in Python 3.x version. 

First, install the sudo apt installation ros-$ROS_DISTRO-ros-numpy and pipe installation open3d

If you go to the https://github.com/HViktorTsoi/FAST_LIO_LOCALIZATION/pull/20 link, you can see the code the code author committed to, and it will run if you change your code with this code.

Thank you.

QingKe-UP, 2024-09-26:
@Parkeunseok Thank you, I changed this code yesterday and fixed the issue. 
Thanks for your reply!